skot/bitcrane #13: FT4232 appears to need an additional pin connected for I2C clock stretching mode

skot, 2024-06-17:
We need clock stretching mode to communicate with the PIC on S19j and earlier hashboards.. According to the [pyftdi docs](https://eblot.github.io/pyftdi/api/i2c.html#pyftdi.i2c.I2cController), the FT4232:
```
SCK should be connected to A*BUS0, and A*BUS7 if clock stretching mode is enabled
```
[image]

This should be U1 pins 16 & 24 pins connected. Ideally this can be hacked on a Bitcrane v2.3 manually to test if it indeed works.

lef149, 2024-06-18:
Clock streching[](https://eblot.github.io/pyftdi/api/i2c.html#clock-streching)
Clock stretching is supported through a hack that re-uses the JTAG adaptative clock mode designed for ARM devices. FTDI HW drives SCL on AD0 (BD0), and samples the SCL line on : the 8th pin of a port AD7 (BD7).

When a FTDI device without an open collector capability is used (FT2232H, FT4232H, FT4232HA) the current sourced from AD0 may prevent proper sampling ofthe SCL line when the slave attempts to strech the clock. It is therefore recommended to add a low forward voltage drop diode to AD0 to prevent AD0 to source current to the SCL bus. See the wiring section.

lef149, 2024-06-21:
Connected pins 16 and 24 and was able to get clock stretching to work and read the pic chip. you do have to play with delays but it seems like you have to no matter what. I did not have to change the clock speed with the right delays.